TinyTapeout/tt-support-tools #10: Cell By Category links to LS cell library not HD

dlmiles, 2023-04-27:
https://github.com/TinyTapeout/tt-support-tools/blob/tt03/reports.py#L8

This links to LS not HD cells.

mattvenn, 2023-04-28:
thanks for pointing that out. It looks like there is no public list of hd cells. Maybe we should have our own.

dlmiles, 2023-04-28:
LS:

https://antmicro-skywater-pdk-docs.readthedocs.io/en/test-submodules-in-rtd/contents/libraries/sky130_fd_sc_ls/cells/

HD:

https://antmicro-skywater-pdk-docs.readthedocs.io/en/test-submodules-in-rtd/contents/libraries/sky130_fd_sc_hd/README.html
https://antmicro-skywater-pdk-docs.readthedocs.io/en/test-submodules-in-rtd/contents/libraries/sky130_fd_sc_hd/cells/inv/README.html

Not that the data appears any different :(   I am working on this what data sheets.

dlmiles, 2023-04-28:
[image]

Sample preview so far.  Working on liberty file data source ATM.  But I think I am still a few months away from first release, then it needs some kind of review (by someone who knows what they are doing).

mattvenn, 2023-05-02:
very cool!
Let me know when you want a review!

mattvenn, 2023-05-31:
see #12 - both are 404 now

closed 2023-06-01 by mattvenn