TinyTapeout/tt-support-tools #14: Investigate why designs with missing ios pass checks

urish, 2023-05-27:
See https://github.com/TinyTapeout/tinytapeout-03p5/issues/3#issuecomment-1565335722 for an example. It has no `uo_out`, and it passes the GDS action successfully (but still caught when fetching user designs by configure.py):

```verilog
module tt_um_TrainLED2_top(
  input wire [7:0] ui_in, // Dedicated inputs
  output wire [7:0] ui_out, // Dedicated outputs
  input wire [7:0] uio_in, // IOs: Input path
  output wire [7:0] uio_out, // IOs: Output path
  output wire [7:0] uio_oe, // IOs: Output enable path (active high 0=input, 1=output)
  input wire ena,
  input wire clk,
  input wire rst_n
);
```

mattvenn, 2023-06-01:
It's never been checked at the project build level. We can enable it, but to do so we'd also need to either change the check to use the yosys inside the docker, or add oss-cad-suite to the gds build.

algofoogle, 2023-06-02:
FWIW, I also noticed that I mistakenly declared `uio_in` as **outputs** and it didn't get blocked other than as a Synthesis Warning summary:
```
Warning: Wire tt_um_algofoogle_solo_squash.\uio_out [0] is used but has no driver.
```

urish, 2023-06-25:
Fixed in tt04 branch

closed 2023-06-25 by urish