TinyTapeout/tt-support-tools #34: No warning if additional clocks aren't configured

MichaelBell, 2024-04-07:
It's not uncommon for people to think that they can clock registers from generated clocks or different input pins, without adding the required configuration to the SDC file.

Currently, no obvious warnings are flagged if you do this, but it is likely your design won't work due to hold violations (setup violations might also be an issue, but would generally not be design breaking as you can just clock more slowly).  This is because STA doesn't understand the clocks you are using so doesn't know to check the paths between different registers in your addition clock domains.

I think this problem will be picked up by there being "unclocked" registers in the STA report.  So it might be a good idea to grep for
`Warning: There are \d* unclocked register/latch pins.`
in multi_corner_sta.checks.rpt and flag a warning on the action summary if this is present.

urish, 2024-04-08:
I've tested on it on `tt06-factory-test`, but I'm getting a warning:

```
===========================================================================
check_setup -verbose -unconstrained_endpoints -multiple_clock -no_clock -no_input_delay -loops -generated_clocks
===========================================================================
Warning: There is 1 unconstrained endpoint.
  _085_/D
```

Here is the relevant excerpt from the GL netlist:
```verilog
  sky130_fd_sc_hd__conb_1 _094_ (
    .HI(_034_)
  );

  sky130_fd_sc_hd__dfrtp_2 _085_ (
    .CLK(clk),
    .D(_034_),
    .Q(rst_n_i),
    .RESET_B(rst_n)
  );
```

and relevant RTL:
```verilog
	reg rst_n_i;
	reg [7:0] cnt;

	always @(posedge clk or negedge rst_n)
		if (~rst_n)
			rst_n_i <= 1'b0;
		else
			rst_n_i <= 1'b1;
```

Do you see any potential issue here?

MichaelBell, 2024-04-08:
Ah, I think think that's because of the async reset.  That's pretty normal so shouldn't be flagged - maybe restrict to startswith "Warning:" and contains "clock"?

urish, 2024-04-08:
Something along these lines?

```python
        sta_checks = "runs/wokwi/reports/signoff/30-sta-rcx_nom/multi_corner_sta.checks.rpt"
        with open(os.path.join(self.local_dir, sta_checks)) as f:
            for line in f.readlines():
                if line.startswith("Warning:") and 'clock' in line:
                    warnings.append(line.strip())
```

Also, any idea where we would find this report in the outputs of OpenLane 2 run?

here's an example of such recent run: https://github.com/TinyTapeout/tt06-dffram-example/actions/runs/8591971841

MichaelBell, 2024-04-08:
Looks good.

In OL2 it looks like this is it: runs\wokwi\52-openroad-stapostpnr\nom_tt_025C_1v80\checks.rpt

closed 2024-04-08 by urish