TinyTapeout/tt-support-tools #77: Missing requirements.txt dependencies: apt-get install python3-tk

dlmiles, 2024-11-03:
`apt-get install -y python3-tk` maybe needed for config_utils.py:6 `import tkinter`

```
(venv) root@a9742d2c0b29:/tmp/ttihp0p2-muldiv8# ./tt/tt_tool.py --check-docs
Traceback (most recent call last):
  File "/tmp/ttihp0p2-muldiv8/./tt/tt_tool.py", line 6, in <module>
    from project import Project
  File "/tmp/ttihp0p2-muldiv8/tt/project.py", line 21, in <module>
    from config_utils import read_config, write_config
  File "/tmp/ttihp0p2-muldiv8/tt/config_utils.py", line 6, in <module>
    import tkinter
ModuleNotFoundError: No module named 'tkinter'
```

urish, 2024-11-03:
Thanks, we're planning remove this (support for config.tcl parsing) for tt10

urish, 2024-11-03:
@htfab assigning to you - please remove config.tcl support after the dust settles for tt09

PR #83 (2024-12-05): fix!: remove deprecated config.tcl support
resolves #77
closed 2024-12-05 by urish